aderusha/HASwitchPlate #140: Some pages not included in demo

proasnet, 2021-03-07:
Hi, I have a six working hasps in my home. Now I am using for RGBWW lights only page 1 on all hasps. Is possible use page 6 as the next scenes buttons, for more colors?
Is any demo, or some example code for yaml file for page 6? Now in their yaml is only automations for buttons, probably for their own colors etc, but not triggers for light scenes or devices.
Thank you very much for any help.

aderusha, 2021-03-07:
Sure!  First, I'd encourage you to check out the latest release [over here](https://github.com/HASwitchPlate/HASPone) (note the new repo).  The HASPone release removes all of the packages and yaml, and replaces it with blueprints.  If you want to have a button on page 6 trigger a scene, after you've [upgraded to HASPone](https://github.com/HASwitchPlate/HASPone/wiki/Updating-legacy-HASP-to-HASPone) you can deploy the [Perform Action](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Perform_Action.yaml) blueprint which will allow you to select a scene to activate with a given button.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha