aderusha/HASwitchPlate #143: Esphome + Nextion mqtt

hassiomanuel90, 2021-05-14:
esphome communicates api with nextion but I would like to use mqtt commands. I have activated mqtt in esphome and when I press a button on the display I get the command with the button identifier. all perfect. The problem is the command to send mqtt to write text on the display. Can I send a mqtt command avoiding the package file you created? thank you

aderusha, 2021-08-04:
First thing, you'll probably want to [head over to the new repository for the project](https://github.com/HASwitchPlate/HASPone) as a lot has changed since this repo was moved over there.

I'm not exactly sure how this interacts with ESPhome, as that is a separate project.  Can you explain a bit more about what you're doing, what you expect, and what is happening?

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha