aderusha/HASwitchPlate #144: stuck on HASP initializing

pljda, 2021-08-04:
Hi, I have the same problem than #80 , could connect once. The stuck on the HASP initializing screen !!
I tested the link between D4 and RXD, D7 and TXD, 5v& Gnd - all are OK.
reinserted the micro SD card - reboot - HASP Initializing ...

Is there a way or bin file so I can reset the screen ?
Thanks - Philippe

aderusha, 2021-08-04:
First thing, you'll probably want to [head over to the new repository for the project](https://github.com/HASwitchPlate/HASPone) as a lot has changed since this repo was moved over there.

If you aren't getting a white screen showing the LCD update process when you power up with a microSD card inserted, [check through this guide](https://nextion.tech/faq-items/using-nextion-microsd/) to ensure that the microSD card has been setup correctly for use with Nextion.

pljda, 2021-08-04:
Hi Allen,
Thanks for your prompt reply. I finally mange to solve my concern with a full retart from the Wemos Mini.
It took me sometime to navigate and find some recap info .
Anyway congrat for your great job.

closed 2021-08-08 by aderusha